theAgingApprentice/HexaFloorRide #18: Audit our use of include files

theAgingApprentice, 2022-11-13:


theAgingApprentice, 2022-11-20:
Doug has started a spreadsheet to track this  but it is a rather convoluted structure.

nerdoug, 2022-12-24:
This topic is closely connected with how we use header files, which is discussed in issue 43. I'll try to keep them separate, at least for now.

Here are some starting points / observations / requirements:
- our explorations into putting our routines into libraries resulted in a complex nested directory structure for parts of the source code. This makes it difficult to group related routines together, and difficult to find any particular routine. Since we've decided not to go with the library approach we can simplify this with 2 principles:
    - related routines will be  grouped together within a single container file for the group, which I'll call a bag to avoid using terms that have other meaning. Actually, there will probably be 2 files for a bag, a .CPP file, and a .H file
    - we'll use a flat directory structure with 2 levels. The first will consist of the src, lib, and include directories. Within these, there will be a single level of what I've referred to as bag files.
- similarly, our explorations of object oriented programming have not convinced us that it's worthwhile to use OOP. Thus we can remove the complexities it brings, and replace the OOP code with equivalent procedural programming code. (I'm using programming classifications from https://www.coursera.org/articles/types-programming-language ) 
- It is important that we continue to create software components that are reusable. Bag files should be completely portable (although I haven't thought through the .H file usage implications yet.) This may mean that you end up compiling and devoting memory to a group of routines when you only need one or two. I think libraries work this was as well, so it must be tolerable. However, we can construct the bag files in such a way that, if required, you can extract a small number of routines. This requires partitioning the .CPP and .H files for the bag so that each has a section for each routine. This needs more thought (recursive includes, shared data structures, automated bag building from atomic routines...)

-------------------------------------------
The current high level structure of our code is:

Our main.cpp structure:
include main.h // this is the only include statement in main.cpp
void setup() with lots of routine calls
void loop() with a few routine calls

Our main.h structure:
include guard
include statements for .h files
global variable declarations + definitions, some const
macro definitions via #define
structure declarations
aaWebService localWebService(WEB_APP_TITLE) call/instantiation ??
function declarations for functions in the include files
includes for 2 pinout .h files
includes for 10 .cpp files
declares for setup and loop

Example mqttBroker.h:
- couldn’t find it. implies all calls to these routines have to come after the routines themselves, risky in the long term.

Example mqttBroker.cpp:
include guard
include main.h // ?? seems strange if it’s included at top of main.cpp
many routines
some embedded variable declaration + definition

include directory:
- flat directory structure
- many .CPP files
- some .H files

lib directory:
- multiple layers of directories
- many duplicated subdirectories
- lots of OOP overhead
- some non code files, xml, yml, python, md documentation, JSON, .txt, .png, .svg ...
- .CPP and .H files are usually paired

Maybe we should just remove all of our stuff from /lib, and convert it to bags in /include? What about case where we had to hack someone else's library?

nerdoug, 2023-02-09:
Am going to experiment with converting the aaChip library to functional programming. It seems to boil down to one 8 line function that is called one time at startup, so it's a good trial candidate for conversion.

Initial survey: 

/lib/aaChip-1.0.1 has
-7 folders in 2 layers
-12 files
-one function in src/aaChip.cpp:

void aaChip::cfgToConsole()
{
   Serial.println("<aaChip::cfgToConsole> APP_CPU (core 1) settings:");  
   Serial.print("<aaChip::cfgToConsole> ... Chip revision = "); Serial.println(getChipRevision()); 
   Serial.print("<aaChip::cfgToConsole> ... SDK version = "); Serial.println(ESP.getSdkVersion()); 
   Serial.print("<aaChip::cfgToConsole> ... Sketch size = "); Serial.print(getCodeSize()); Serial.println(" bytes");  
   Serial.print("<aaChip::cfgToConsole> ... Free heap = "); Serial.print(getFreeHeap()); Serial.println(" bytes"); 
   Serial.print("<aaChip::cfgToConsole> ... Serial baud rate = "); Serial.print(getSerialSpeed()); Serial.println(" Hz");
   Serial.print("<aaChip::cfgToConsole> ... Arduino core = "); Serial.println(getCpuId());
   Serial.print("<aaChip::cfgToConsole> ... Arduino core clock frequency = "); Serial.print(getCpuClock()); Serial.println(" MHz");
} //aaChip::cfgToConsole()

- startup console output before conversion:
...
H) showCfgDetails{5}>  Robot Configuration Report
H) showCfgDetails{5}>  ==========================
<aaChip::cfgToConsole> APP_CPU (core 1) settings:
<aaChip::cfgToConsole> ... Chip revision = 3
<aaChip::cfgToConsole> ... SDK version = v4.4.2
<aaChip::cfgToConsole> ... Sketch size = 1151776 bytes
<aaChip::cfgToConsole> ... Free heap = 172736 bytes
<aaChip::cfgToConsole> ... Serial baud rate = 115201 Hz
<aaChip::cfgToConsole> ... Arduino core = 1
<aaChip::cfgToConsole> ... Arduino core clock frequency = 240 MHz
V: <showCfgDetails> Network connection status = TRUE

I moved the routine 
- from /lib/aaChip01.0.1/src/aaChip.cpp...void aaChip::cfgToConsole()
- to /include/configDetails.cpp...void cfgToConsole()  // migrated from /lib/aaChip-1.0.1
- removed "aaChip:: from the source identification string
- added void cfgToConsole() to global_variables.cpp
- changed call to cfgToConsole in showCfgDetails
- removed aaChip appCpu instantiation from main.cpp
- removed #include <aaChip.h> from main.cpp
- backed up, then removed /lib/aaChip-1.0.1
- compiled without warnings or errors

post conversion startup console output:

H) showCfgDetails{5}>  Robot Configuration Report
H) showCfgDetails{5}>  ==========================
<cfgToConsole> APP_CPU (core 1) settings:
<cfgToConsole> ... Chip revision = 3
<cfgToConsole> ... SDK version = v4.4.2
<cfgToConsole> ... Sketch size = 1151456 bytes
<cfgToConsole> ... Free heap = 172656 bytes
<cfgToConsole> ... Serial baud rate = 115201 Hz
<cfgToConsole> ... Arduino core = 1
<cfgToConsole> ... Arduino core clock frequency = 240 MHz
V: <showCfgDetails> Network connection status = TRUE

about to Git merge branch issue90g (poorly named - should be issue 18)  to complete this.

nerdoug, 2023-02-09:
git merge completed. /lib/aaChip-1.0.1 is no longer in use, This was an easy task, and if other libraries are like this, the conversions won't take long.

Interesting that the information displayed includes the sketch size, so we can see that the conversion kept the same functionality, but saved 320 bytes.

nerdoug, 2023-02-12:
Guess I'll follow alphabetical order, and do aaFlash next. It looks t be as easy as aaChip was.