theAgingApprentice/HexaFloorRide #5: Decide on connectors

theAgingApprentice, 2022-11-13:
The connector selected must satisfy this criteria:

1. Must handle be 6 wires.
2. Be an industry standard interface. We do not want to to have to make fiddly error prone ends. 
3. Must handle up to 10 amps power  for occasional bursts of motor draw.

nerdoug, 2022-11-14:
Will create a diagram showing a proposed configuration of PCB's, cables, and connectors required. Once we have the requirements figured out we can decide on how we implement them.

Quick notes:

- servo driver board has 
    - header pins for most signals
    - wire terminal block for 5V and its Ground
 - normal trace width will suffice for everything except 25V and 5V and their ground paths
 - for 25V and 5V wires, suggest using stranded copper wire, some terminal blocks and some yet to be determined connectors
 - connectors should lock into place, so vibration doesn't cause them to come loose
 - not sure if initial testing should be done based on breadboard, or soldered proto-board

nerdoug, 2022-11-16:
a reference on ways to implement current limiting:
[https://www.electronics-notehttps://www.electronics-notes.com/articles/analogue_circuits/power-supply-electronics/current-limiter-circuit.php#](https://www.electronics-notehttps://www.electronics-notes.com/articles/analogue_circuits/power-supply-electronics/current-limiter-circuit.php#)

Looked at current limiting diodes on Digikey. The available ones either have Amp values below 0.2 A, or voltages below 5 V.

An interesting approach might be to use a 20V to 5V regulator that can limit the 5V current. I think this is a feature of many regulators.

nerdoug, 2022-11-16:
not a great choice, but interesting: [https://www.ebay.com/itm/262459427526](ebay/china)

Another: [https://www.amazon.com/Diymore-Adjustable-Regulator-Converter-Constant/dp/B07Y7YB14L?th=1](Amazon) This one is beefy so we wouldn't be pushing it hard, and it has current control, via multi-turn resistor. We could set it to the highest current our circuit can handle, and it will block any attempt to draw more current.