theAgingApprentice/HexaFloorRide #8: Decide on wire guage required for motor power

theAgingApprentice, 2022-11-13:


theAgingApprentice, 2022-11-14:
Here are notes on what we need for wire gauge. If we need to handle 15 Amps then that required 14 gauge wire which is household wiring. I do not think that this makes any sense and seems way over kill.

1. cat5 is a 24AWG rated cable. This means it’s a 24 gauge wire. 
2. As the gauge increases, the diameter of the cable decreases. So does the current capacity. 
3. For 24 AWG copper insulated cables, the ideal current is from 2.1 Amp to 3.5 Amp. 
4. The calculation for the maximum capacity is:    Cat5 Current Capacity = 4 X Size of Cable in Sq.mm
                                          = 4 X 0.88 Sq.mm
                                          = 3.52 Amps

Amperage Capacities for Standard Non-Metallic (NM) Cable
--
16-gauge wire | 13 amps
14-gauge wire | 15 amps
12-gauge wire | 20 amps
10-gauge wire | 30 amps
8-gauge wire | 40 amps
6-gauge wire | 55 amps
4-gauge wire | 70 amps
3-gauge wire | 85 amps
2-gauge wire | 95 amps

Wire Use | Rated Ampacity | Wire Gauge
Low-voltage lighting and lamp cords | 10 amps | 18-gauge
Extension cords (light-duty) | 13 amps | 16-gauge
Light fixtures, lamps, lighting circuits | 15 amps | 14-gauge
Kitchen, bathroom, and outdoor receptacles (outlets); 120-volt air conditioners | 20 amps | 12-gauge
Electric clothes dryers, 240-volt window air conditioners, electric water heaters, sometimes cook tops and wall ovens | 30 amps | 10-gauge
Cook tops and ranges | 40-50 amps | 6-gauge
Electric furnaces, large electric heaters | 60 amps | 4-gauge

nerdoug, 2023-01-10:
issue 71 is related to this one, and may be caused by motor power wiring not being robust enough in our prototypes.

nerdoug, 2023-01-13:
There are 2 sides to power wiring capacity. One is the ability to carry the required current (which we're not entire sure of yet) without an unacceptable heat build up. The second is having a low enough resistance so that the voltage that gets to the servo controllers isn't reduced below the minimum of 4.8 volts, notably when the worst case current is being carried. The second requirement is illustrated in git/.../pcb/power/Servo-power-wiring.pdf. To resolve the voltage drop, I used 18 AWG stranded steel wire which completely removed the previous voltage drop. It may be more than is needed, but we'll need to figure out what the max current we need to handle is to determine the needed wire gauge and circuit trace widths. It may turn out that we can't use pcb traces for the 5V power, and need to use wires.