theAgingApprentice/HexaFloorRide #43: Porpose standard for header file usage

theAgingApprentice, 2022-11-20:


nerdoug, 2022-12-23:
Some notes that haven't been formed into a proposal yet...

Header file standard considerations

- routine syntax must be defined before all calls, notably if call is before the routine’s code
- therefore routine declaration needs to be in a common header file that’s encountered before any code that uses it. Preferably before any code at all, to simplify and reduce risk of indirect change induced problems
- don’t want a large number of small files – can’t find anything
- group routines together by functional role. Need a new word – call this a bag of routines
- each bag has a .cpp file, and a .h file
- each routine has a section in cpp file, and in .h file
- only need one include guard per bag, not per routine
- we don’t use namespaces or classes or templates 
- we don’t use inline code, but maybe we should. Inline function definitions go in .h files

From: https://learn.microsoft.com/en-us/cpp/cpp/header-files-cpp?view=msvc-170

-what to put in header files
    • include guard (at bag level)
    • syntax declarations for variables, functions, classes
    • inline functions

-what NOT to put in header files
    • built-in type definitions at namespace or global scope
    • non-inline function definitions
    • non-const variable definitions
    • aggregate definitions
    • unnamed namespaces
    • using directives

Declaring variables only at the top of the function is considered bad practice. Place declarations in the most local blocks.
Defining variables ( var = 12  vs  int var)
- at top of main function
- in routine that first uses it, or routine that mainly uses it
- not in a header file, only in cpp

code structure
-global variables (considered bad practice)
-includes

Need to analyse .H file implications of removing unneeded object oriented complexity.

https://stackoverflow.com/questions/28517555/c-declaration-of-class-variables-in-header-or-cpp

"variable declarations are not supposed to be in the header file".... what? The member variables must be declared with the class declaration. 
      – [Cory Kramer](https://stackoverflow.com/users/2296458/cory-kramer)
       [Feb 14, 2015 at 16:20](https://stackoverflow.com/questions/28517555/c-declaration-of-class-variables-in-header-or-cpp#comment45352263_28517555)
    • If the variable is meant to be used globally, then extern goes into the header, and you define the variable once in some .cpp file. If they're global variables used internally, then they go into the .cpp. If they're logically part of the class, then make them member variables. 
      – user3920237
       [Feb 14, 2015 at 16:21](https://stackoverflow.com/questions/28517555/c-declaration-of-class-variables-in-header-or-cpp#comment45352276_28517555)
    • Variable declarations and member variable declarations are different. And besides, it's the definition of non-member variables that will give you problems, not the declarations. 
      – [Some programmer dude](https://stackoverflow.com/users/440558/some-programmer-dude)
       [Feb 14, 2015 at 16:22](https://stackoverflow.com/questions/28517555/c-declaration-of-class-variables-in-header-or-cpp#comment45352290_28517555) 

Doug: our code only has one compiled module, main.cpp (with a bunch of includes) so we don’t need to use extern.

Our main.cpp structure:
include main.h   // this is the only include statement in main.cpp
void setup() with lots of routine calls
void loop() with a few routine calls

Our main.h structure
include guard
include statements for .h files
global variable declarations + definitions, some const
macro definitions via #define
structure declarations
aaWebService localWebService(WEB_APP_TITLE)  call/instantiation ??
function declarations for functions in the include files
includes for 2 pinout .h files
includes for 10 .cpp files
declares for setup and loop

Example mqttBroker.h
(couldn’t find it)

Example mqttBroker.cpp
include guard
include main.h  // ?? seems strange if it’s included at top of main.cpp
many routines
some embedded variable declaration + definition

nerdoug, 2023-01-15:
more stuff to ponder: from https://www.acodersjourney.com/top-10-c-header-file-mistakes-and-how-to-fix-them/
Mistake # 1: Not using “include guards” in a header file.
MISTAKE # 2: Incorporating "using namespace" statements at top level in a header file
MISTAKE # 3 : Having multiple unrelated functionality grouped together in a single header file (and cpp file)
MISTAKE # 4: Not making the header file compilable by itself
MISTAKE 5.a : Including non-required header files in your header – for example, including files that only the .cpp file code needs .
Mistake # 5.b : Putting too much information in a header file and causing information leakage.
Mistake # 6: Not explicitly including all STL headers required by your cpp code file.
Mistake # 7: Not making judicious use of forward declarations in header files
Mistake # 8: Including a cpp file in a header file.
Mistake # 9: Declaring functions shared between multiple cpp files in separate header files/ code files.
Mistake # 10: Putting your project’s header files into the precompiled header file.

nerdoug, 2023-01-15:
Made a bunch of changes under this branch/issue that probably should have been under 18:

in main.cpp:
- remove call to setupMobility(), which is part of Andrew's old leg movement  code

in main.h
- noted that there's a lot of stuff in here that probably shouldn't be in a header file, but for now, just cleanup
- remove //#include <InverseK.h> which is an inverse kinematics library we never used
- remove parameters for Andrew's old leg movement stuff
- remove #include <ikLibrary.cpp> - an unused inverse kinematics library
- remove #include <servoLegs.cpp>Andrews unused leg movement code

in oled.cpp
- remove command to display info from Andrews old leg movement code on OLED.

nerdoug, 2023-01-15:
this issue closed when I uploaded code cleanup so I reopened it.

nerdoug, 2023-01-15:
Here's a first draft of a proposed header file usage standard:

- gathered from various websites, and by observation of Adafruit libraries, etc.
- bag is a collection of related functions, with a header section and code section
     - header section has a sub-section for every function
     - code section has a subsection for every function

- header files or sub-sections can include:

     - syntax declarations for function calls within the associated cpp file or bag (FOB) that are used by other FOBs
     - syntax declarations for global variables that are const
     - syntax definitions for classes & templates
     - macro definitions via #define statement
     - inline function definitions

- header files can not contain:
      - non-inline function definitions
      - non-const variable definitions
      - aggregate definitions
      - unnamed namespaces
      - using directives

Our code has only one compiled module, main.cpp (with a bunch of embedded include files) so we don’t need to use extern. Libraries are a separate issue.

We can put include guards in our header files, but they shouldn’t be needed: main.cpp includes main.h, and main.h includes all other header files. Each header file gets processed exactly once. No file other than main.h should include any other header file.

Our main.h file has a lot of stuff that isn’t allowed according to the above conventions, which we should discuss and/or change.

Implications:
- where do non-const global variables go?
     - Declaration in main.h  Eg:  float x_tilt_angle
     - Definition in main.cpp, probably right after include main.h and before any executable code. Eg:  x_tilt_angle = 0.0
- could put non-const global variables in a separate include file to keep main.cpp tidy
- because all header files are encountered by the compiler before any executable code, functions can appear in any order, but need a declaration in their header file.
- order of compiler processing:
        initially processing main.cpp
        does main.h
              it does all other .h files
        does all non-const global variable declarations and definitions, perhaps via an include file, say global_variables.cpp
        does all includes of executable code, perhaps via an include file, say all_includes.cpp
        does executable code in main.cpp

Might be simpler to put both const and non-const global variables together in the function's section in main.h and main.cpp?

commit 96a21919 (2023-01-15): cleanup unused files
Fixes #43
PR #91 (2023-01-15): Propose standard for header file usage
Fixes #43
commit c734ea0a (2023-01-15): Propose standard for header file usage
Fixes #43
closed 2023-01-15 by nerdoug