theAgingApprentice/HexaFloorRide #44: Buy voltage regulator

theAgingApprentice, 2022-12-11:
See  details of which regulator to purchase in the comments of issue #5

nerdoug, 2022-12-12:
4 voltage regulators ordered from Amazon with projected delivery on Dec 24.

part description: 
https://www.amazon.com/Diymore-Adjustable-Regulator-Converter-Constant/dp/B07Y7YB14L?th=1

Order:
Order [#113-0437045-2593032](https://www.amazon.com/gp/r.html?C=1GDZONJ9HF37K&K=6GDADTOYDXIL&M=urn:rtn:msg:20221212222112d6f4d6137a1e43bb960e14faa480p0na&R=3DSA7YIZWUWJ5&T=C&U=https%3A%2F%2Fwww.amazon.com%2Fgp%2Fcss%2Forder-details%3ForderId%3D113-0437045-2593032%26ref_%3Dpe_386300_440135490_TE_simp_od&H=GA1BWEZQQEVLJ8A5ETNTUIQ9DFMA&ref_=pe_386300_440135490_TE_simp_od)

Not that a 5V > 3.3 V regulator is also needed but it is low current, and easy to find.

nerdoug, 2022-12-21:
Four of these voltage regulators have arrived. We should probably test them and maybe do some documentation, like how to adjust output voltage and current limit (i.e which way to turn the multi-turn pot screws.)
Leaving this issue open until testing and documentation done.

nerdoug, 2022-12-22:
uploading odt and pdf versions of file voltageRegulator-20A to pcb/power in github. It contains documentation and calibration procedures.

PR #55 (2022-12-22): Buy voltage regulator
document 20A vReg, and calibration procedure
Fixes #44
commit 89fc074e (2022-12-22): Buy voltage regulator
Fixes #44
closed 2022-12-22 by nerdoug