theAgingApprentice/HexaFloorRide #49: The address 192.168.2.158 does not display correctly

theAgingApprentice, 2022-12-18:
In the module aaNetwork.cpp in the routine cfgToConsole() the display of the robot IP address is sometime incorrect. Specific use case was the address being assigned to Doug's robot on Andrew's network, 192.168.2.158 was displayed as 0.104.64.63. We put in a debug line that help ed uncover this issue an the code that actually worked was WiFi.localIP().toString(); This can be displayed with far less work and is correct. The code _convert.ipToString(WiFi.localIP().cStr());

theAgingApprentice, 2022-12-18:
The issue just popped up on the Andrew robot. Here is what led to this:
1. I closed the current terminal connected to a working ESP32
2.  started a new terminal session 
3. I rebooted the ESP32 
4. The trace output from the ESP32 showed up without needing to rebuild/upload the code
5. I then saw not only the wrong IP address, but the same incorrect IP address as we saw on Doug's robot.
6. I then rebuilt/posted to the ESP32
7. The output returned to normal 

From this experience we can conclude that the value of the IP address is not relevant to the bug. It must be some other subtle thing. We seem to have a good workaround (see last note on this issue) and may never know the root cause of the bug.

nerdoug, 2022-12-19:
It seems that WiFi.localIP() is provided as a string, and no conversion is needed.

theAgingApprentice, 2022-12-20:
Would be  great to reduce the complexity of the code as you have discovered is possible. The MAC address of the ESP32 that you loaned e is 94B97E5F52DC.  Also,  while you are in  there messing around you should change the prefix of the robot name from Hexbot to HexaFloorRide.  For example, in MQTT my robot current registers as Hexbot94B97E5F52DC but he really should call himself HexaFloorRide94B97E5F52DC.

nerdoug, 2022-12-20:
I'm about to wrap up this issue, following your suggestions. A few philosophical thoughts first:
- we're using the ESP32 MAC address to identify the robot chassis, for example to identify soft servo offsets.
- if we swap ESP32's that falls apart
- it would be nice to have a real chassis ID, which could be:
      - some DIP switches
      - a 4 bit rotary switch
      - an I2C readable value, maybe something in PCA9685?
 - as we already discussed, using NV RAM in ESP32 for MQTT Broker IP address has similar problems
      - could scan through a list of possible broker IPs, but that doesn't work in new demo environment
      - could use a symbolic name, and look it up in DNS, if DHCP can supply a DNS service pointer
      - might be able to do a MAC level broadcast, that invites Broker to connect and identify itself
      - might also be able to do that at IP layer, maybe more easily
 
OK, into the code.

theAgingApprentice, 2022-12-21:
We could adopt an idea that will take work and time but is possible. We can put unique serial numbers on each chassis. When the robot looks at this memory and it is not set it could use the current MAC address scheme. And issue a please set my serial number value message to the MQTT broker. If the serial dumber is set then use that instead of the old scheme. To be really ambitious, down the road if we add vision the robot could scan its own serial number and set its name for itself. Well worth discussing further I think.

theAgingApprentice, 2022-12-21:
After re-reading your comment and mine here is a basic question to resolve. What is the essence of the robot identity?

1. Your suggestion implies that the PCB (circuit or brain) is the answer.
2. My suggestion implies that the chassis (the body) is the answer.

Either way can work. I guess we have to do some robot philosophy to decide which way we want to go. Truly this hobby copies every area of the sciences!

theAgingApprentice, 2022-12-21:
I think this naming scheme is really a new topic. I have created issue #53 to track the idea of a new naming scheme for the robot. Let’s keep this ticket focused on fixing the software bug that causes the wrong IP address to be displayed. This will ensure that each issue only tracks one item and can be closed. Nothing worse then an issue that needs to stick around forever and has to be renamed as the topic shifts. Over time this approach tens to make it hard to find old tickets I think.

PR #54 (2022-12-21): fix console IP display, host name prefix, have2 MACs / house
fix console IP display, host name prefix, recognize 2 MAC's per house
Fixes #49
commit 537d768f (2022-12-21): fix console IP display, host name prefix, have2 MACs / house
Fixes #49
closed 2022-12-21 by nerdoug