theAgingApprentice/HexaFloorRide #90: Create a print macro for our code

theAgingApprentice, 2023-01-15:


nerdoug, 2023-01-16:
about to put a design document trace-design.pdf * .odt  into /docs, which will close this issue
I changes the title of this issue to DESIGN a trace message facility, to be followed up by another titled IMPLEMENT...

theAgingApprentice, 2023-01-29:
After reviewing the current design we have new ideas and Doug will try to simplify the design a little.

nerdoug, 2023-02-01:
looks like an issue can be in progress, but closed, which makes it disappear altogether from VSC's git system. I've reopened this one

nerdoug, 2023-02-03:
Simplified, but not function complete tracing is working, with support in these places:

global_variables.cpp: symbols and control table allocation (search for t$H)
configDetails.cpp:  run time initialization of control table (search for setupTracing()
terminal.cpp:  the tracer routine that decides if a requested trace is enabled, and if so generates it (search for void tracer)
flows.cpp: some test traces. search for trace(
/docs/trace-design.odt,pdf:  descriptive documentation of trace facility

The big impact is that we get compiler warnings (eventually suppressed) for each routine that has tracing enabled, because the routine name and identifier number are remembered as macros (to avoid generating any executable code) which are redefined for each routine.

The preparation at the beginning of each routine that has trace commands is:
     #define localRName "setupFlows"       //name of the routine as it will appear in the trace message
     #define localRNum 3                           // the numeric identifier for this routine

The way you invoke a trace message is
     trace(t$E, "message/label", variable name)     where t$E represents the trace type, t$E means error
example:
     trace(t$M,"entered routine", 0)

The trace output looks like this on the console:
     M) setupFlows[3]:entered routine 0.00

This is implemented in merged branch issue90b, except that documentation isn't complete yet.

theAgingApprentice, 2023-02-05:
We had a discussion about the trace message facility today. Here is what we discussed that could be added to the design document.

1. Message breadcrumb details that get prefixed to a message are what we refer to as the message **FORMAT**. The message format can include the following details prefixing the message itself:
* File name, accessible via the GCC pre-compiler macro __FILE__
* Function name, accessible via the GCC pre-compiler macro __FUNCTION__ or __func__
* Line number, accessible via the GCC pre-compiler macro __LINE__
2. We use five **TYPES**. Those types are
* Info high
* info medium
* info low 
* warning
* error
3. The behaviours of messages that we can be controlled include. 
* what type of messages to display. 
* what format the messages should have. 
5. The **SCOPE** of behaviours are either global or local.
6. There are two points in time when we can determine the behaviour of messages. Those times are **BUILD TIME** and **RUN TIME**.
7. We following truth table shows what the current trace feature is designed to support:
* Trace **TYPE** set at **COMPILE TIME** = YES
* Trace **TYPE** set at **RUN TIME** = YES
* Trace **FORMAT** set at **COMPILE TIME** = YES
* Trace **FORMAT** set at **RUN TIME** = NO

nerdoug, 2023-02-07:
We can't convert messages coming from libraries to the new tracing facility, because they are compiled separately, and don't have access to the tracing symbols, macros or data structures in our code. This is unfortunate, because they have unconditional print statement that account for a fair amount of the console clutter.

Possible approaches:
- do nothing, and accept the uncontrolled console messaging. I'm not in favour of this because I think there's a significant development velocity improvement if we make the console messaging more precise
- implement a simpler control mechanism within the libraries that allows compile time control, via a small number of symbols, of the print statements that are currently unconditional. This might apply to large or complex libraries, but for simple ones, the next approach might be the same amount of work with more benefit.
- convert the libraries to functional code, especially in cases where it's easy, and there's a high logical and source file overhead for a small amount of code. For example the aaChip library is effectively one 8 line routine cfgToConsole() with a bunch of wrapper functions. We probably want to do this migration anyway as part of issue 18
- comment out most of the print statements in libraries that were useful during debugging and development, but now don't really serve a purpose. Easiest to do, and if the messages don't really add value, probably OK, especially if the print commands remain in place, but commented out. A practical approach, but it feels wrong. 

I'm leaning towards migrating libraries to functional code, and will experiment on aaChip under a new issue.

Two git merges on the way:
1) get the latest trace support code into git, including partial conversion of the setup routine to tracing
2) update the trace-design doc, which is my next task.

theAgingApprentice, 2023-02-07:
I think I agree with your approach of migrating the libraries. There is a lot of old code in there that could benefit from being revisited and having a second set of eyes on it. Happy to help with this process.

nerdoug, 2023-02-08:
Just merged branch issue90d which added the bulk of the trace functionality, except for routing tracing to MQTT. Next step is to update the trace-design doc in /docs, then work on the MQTT stuff.

nerdoug, 2023-02-08:
just merged branch issue90e with minor tracing improvements plus full documentation in docs/trace-design.odt/pdf.
remaining: allow traces to go to MQTT topic, and update document to describe this.

theAgingApprentice, 2023-02-26:
This has been fully designed and documented. A separate issue will track  implementing this through out all of our code.

commit 2ef59587 (2023-01-16): Design a trace facility for our code
Fixes #90
closed 2023-02-26 by theAgingApprentice