theAgingApprentice/HexaFloorRide #129: Fix lint errors with new header standard

theAgingApprentice, 2023-02-12:


nerdoug, 2023-02-14:
Today, there are no files in red in VSC. However I did see a case earlier today where a simple "while" statement had red underlines although it was identical to other similar structures in the same source file.  However when I came back to that file (mqttBroker) there were no underlined parts, and no filenames in red. It's almost as if linter takes a long time to  figure things out.

Willleave this open for a while to see if red filenames return.

nerdoug, 2023-02-17:
It seems that Linter is gong to require us to configure our files so each source file can be independently compiled so that Linter has access to all variables macros and includes when we're editing a single source file. This is disappointing, and I suspect will lead us back to the web of reference interconnections we were trying to fix. Disabling Linter is tempting, but probably not acceptable. I'll start reworking the header file standard.

closed 2023-03-11 by nerdoug

nerdoug, 2023-03-11:
- moved some includes from main.cpp to main.h
- put an include main.h in almost all cpp files
- compiles OK
- /lib file have their own .h file references, and don't follow our new conventions. Best to address this as we convert them from OOP to functional programming

PR #178 (2023-03-11): Fix lint errors with new header standard
Fixes #129
commit 33578098 (2023-03-11): Fix lint errors with new header standard
Fixes #129