theAgingApprentice/HexaFloorRide #134: MQTT message side reduction exercise

theAgingApprentice, 2023-02-12:


nerdoug, 2023-02-14:
re using MAC addresses in identifiers:  Representing an 8 bit binary byte by using 2 8 bit ASCII characters to represent the hex digits for 4 bit nibbles isn't very efficient. However, it does guarantee that printable characters are used that aren't likely to be misrepresented or mishandled along a communications path.

There is another approach that maintains printability but is more efficient- Base64. details at:
   https://en.wikipedia.org/wiki/Base64

nerdoug, 2023-02-14:
We seem to have two conflicting definition for HOST_NAME_SIZE:

$ pwd
/c/Users/DougElliott/Documents/VisualStudioCode/C/Projects/HexaFloorRide

$ grep -i -r "host_name_size ="
lib/aaNetwork-1.0.0/src/aaNetwork.h:static const int8_t HOST_NAME_SIZE = 40;
lib/aaWeb-1.0.0/lib/aaNetwork-main/src/aaNetwork.h:static const int8_t HOST_NAME_SIZE = 20;

Second one is in a copy of aaNetwork.h that's inside aaWeb-1.0.0 and probably isn't compiled, as two conflicting const definitions would likely be an error. I'll ignore this, expecting it to become moot once aaWeb is converted to functional programing.

This points out that my efforts to implement the new header standards missed header files that are copies embedded in libraries. Solution is to simplify the libraries, already planned.

nerdoug, 2023-02-14:
changes made to implement unique names as hfrMMMM where MMMM is the hex for the last 2 bytes of the MAC address. This reduces name length from 25 to 7 characters, and still has a high probability of uniqueness.

- in global_variables, change definition of HOST_NAME_PREFIX to "hfr". The lower case contrasts with upper case in MAC address. Note the lack of a leading underscore in variable name, due to object instantiation.
- 
- in aaNetwork.cpp, recode getUniqueName function to create name as the _HOST_NAME_PREFIX plus the hex for the last 2 bytes of the MAC address, with the colon removed.

- in aaMQTT.cpp, changed definintion of TOP_OF_TREE from "agingApprentice/" to "aa/" for brevity

Noticed that MQTT help command stopped working, and uses old unique name.

nerdoug, 2023-02-14:
also noticed that the layout of the beginning of MQTT messages seems to be inconsistent, but that was the case before I started the size reduction effort, so I'm opening a separate issue (136) for that.

PR #137 (2023-02-14): MQTT message side reduction exercise
Fixes #134
commit 0a699241 (2023-02-14): MQTT message side reduction exercise
Fixes #134
closed 2023-02-14 by nerdoug