theAgingApprentice/HexaFloorRide #145: Investigate leg movement being jagged rather than smooth

nerdoug, 2023-02-17:


nerdoug, 2023-02-17:
I converted the console progress messages from serial prints to trace commands so they can be disabled for testing. However, they display only at the end of each leg movement (i.e. flow row) and not at the end of the 50 msec frames we divide each movement into.

I suspect the calculations for each frame are done at the end of the previous frame's movement, and the time it takes to do the calculations acts like a pause between frames. If this is the case, we may be able to rearrange the code so that the calculations are done in parallel with the leg movement, i.e. pre-calculate the numbers before they're needed.

nerdoug, 2023-02-17:
disabling the do_flow traces so there's no serial during leg movement didn't solve the issue.

nerdoug, 2023-03-03:
spawning a related issue: implement some performance metrics to see how much time our software tasks are taking, how consistent they are, and what the average and peak CPU usage is.

nerdoug, 2023-03-04:
cpu metrics are working, and it seems that web monitoring is absorbing 98.5 % of CPU on an idle system. That may be a factor in ragged leg movements. Will know soon, because I think I have an easy fix for the web monitoring issue - only check it every, say, 500 milleseconds, instead of every loop() execution.

nerdoug, 2023-03-13:
I changed the pushup script to create frames more frequently, and the jagged leg movements went away. The defalt is 50 millisecond per frame, and I changed it to 25 and then 20 msec. To my eye, it looks pretty smooth, but a second opinion is needed.
I made these changes by changing options in the DO_FLOW command that is the last command in the script that is sent to the robot. I used:  send("DO_FLOW,1,20")  which says just move the toes (the 1) and use a 20 msec frame interval (the 20).

I actually enabled debug output on every toe movement (i.e. every frame) by using send("DO_FLOW,9,20") at the end of my script, and saw that as expected, a lot more frames are being generated, and the Flow processing is taking up more CPU time. The Flow CPU maxed out around 33 percent, but part of that is the time to generate the massive amount of debug I/O that was going to the console. I captured this to the file 20msecFrames.txt which I will put in git/performance/mqttScripts/. I should do one without the debug traces, just the CPU usage, and see if the flow CPU is lower. That will be 20msecFrames-noDebug.txt.
Too bad I haven't converted Do_Flow to the new dispatching - it would be very interesting to see how often it's getting delayed dispatching when things are this busy.

Much better without the debug output - max of 22%, and mostly around 16%. I'll convert Do_flow to the new measurable dispatch system for better info, and will probably do some optimization in any case.

nerdoug, 2023-03-13:
re-opening until we get better measurements of do_flow performance so we can judge if it's safe to run 20 msec frames.

theAgingApprentice, 2023-03-19:
Re-opening because we still need to examine the CPU cost of a higher frame rate.

theAgingApprentice, 2023-03-19:
With our robots side by side we tested both 20 and 50 msec frame rates to see if the robots behave the same way. Results at 20msec and 50msec is the Doug's robot works find and Andrew's robot loses control of the left side of his body. Possible issues for Andrew's robot

1. The left PCA9685 (I2C address 0x41) has an issue, maybe the 1000uF capacitor?
2. The I2C wiring is flaky
3. The ongoing MQTT issue is interfering somehow

MQTT messages that come up regularly are:
[I][AsyncMqttClient.cpp:674] _sendPing(): PING
[I][AsyncMqttClient.cpp:382] _addBack(): new back #12
[I][AsyncMqttClient.cpp:414] _handleQueue(): snd #12: 2/2
[I][AsyncMqttClient.cpp:424] _handleQueue(): p #12 rel
[I][AsyncMqttClient.cpp:244] _onAck(): ack 2
[I][AsyncMqttClient.cpp:249] _onData(): data rcv (2)
[I][AsyncMqttClient.cpp:267] _onData(): rcv PINGRESP
[I][AsyncMqttClient.cpp:497] _onPingResp(): PINGRESP

PR #193 (2023-03-13): Investigate jagged leg movements
Fixes #145
commit 23042ba1 (2023-03-13): Investigate jagged leg movements
Fixes #145