theAgingApprentice/HexaFloorRide #156: Correct power design

theAgingApprentice, 2023-02-19:


nerdoug, 2023-02-20:
Changes needed:
-ESP32 gets it's power via the micro USB connector. There may be other ways, but they're a bit risky, and we've found that if you make a mistake you can smoke a Huzzah board easily
-need to add a second USB connector for the serial monitor connection, and control how it's connected to ESP32
- data leads can be wired straight from external USB to ESP32 USB
- the +5 and GND leads on the ESP32 USB need to go to a DPDT switch that selects the 5V CPU power source:
* the external USB connector, or
* output of a 20V - 5 V voltage regulator connected to battery power. This is different than the power for servos, which will tend to be less stable, and will likely be higher that 5 V. This will power the ESP32 when it's not connected to the serial monitor
- we decided to use a separate voltage regulator for 3.3 volts rather than use the 3.3 volts provided by the CPU. This will need to work for both positions of the CPU power source selection DPDT switch. 

- need to have fuses/circuit breaker and safety diodes to defend against all possible mis-connections.

nerdoug, 2023-03-13:
clarifying the voltage requirements of the PCA9685:

Servo Voltage - V+
- according to the Adafruit spec for the PCA9685 board, you can put up to 12V through V+, the power that goes to the servos.
- according to the HI-TEK HS485HB servo spec, their specified voltage range is 4.8 to 6 Volts

Logic voltage, Vcc or Vdd
- on the Adafruit pinouts page, they say Vcc should be 3 - 5 V max
- on the Adafruit schematic, Vdd is specified as 2.3 - 5.5 V
- the NXP Semiconductors datasheet for the PCA9685 chip says Vdd should be from 2.3 to 5.5 V in section 12

Recognizing that the 3.3 voltage available on Huzzah32 physical pin 2 comes from an AP2112-3.3 voltage regulator on the Huzzah board, but not part of the CPU, we've decided to use this to source 3.3v rather than add another voltage regulator.

nerdoug, 2023-03-14:
issue addressed in the power design in hfr-power-overview-phase1

PR #195 (2023-03-14): phase 1 power design with control panel & board division
Fixes #156
commit 923d208f (2023-03-14): phase 1 power design with control panel & board division
Fixes #156
closed 2023-03-14 by nerdoug