theAgingApprentice/HexaFloorRide #158: Decide approach on main circuit board

theAgingApprentice, 2023-02-19:
Do we want to re-purpose the old PCB first then make a new one or just go for a new one right away?

nerdoug, 2023-02-20:
The old PCB was designed for TWIPE, and has stepper motor controllers, and an IMU that we don't need, so we probably don't want to use it. Steps for developing circuit support for hfr include:
- deciding what functionality (and which components) will go on which circuit board
- creating prototypes for each using breadboards and/or soldered proto-boards
- get PCB's manufactured and stuffed.

nerdoug, 2023-03-14:
The power design in hfr_power-overview_phase1.pdf has a high level partition of functionality into 3 circuit boards:
   control panel, having all the switches, indicators and connectors that humans need easy access to
   CPU board, with ESP32 and power control circuit
   Voltage regulator board with 2 largish voltage regulators, protection diodes, fuse, etc

Phase 1 doesn't implement the power control system that uses 7050 and 2141 chips. This is deferred to phase 2. It's not clear that all of this will fit on one board, but separating the power control stuff is painful because of the large number of connections to the CPU board that would be needed.

Prototping this approach will go hand in hand with getting battery operation working.

nerdoug, 2023-03-21:
The functions included in the main, or processor or CPU board are documented in the file:
      /pcb/power/hfr-power-overview-phase1.pdf
and include:
- Huzzah32 CPU board
- battery monitor
- sensor connections
- servo power control (7050 and 2141 chip stuff)

as well as connectors for possible future expansion:
- I2C bus
- SPI bus
- 1 Wire bus

closing this issue

closed 2023-03-21 by nerdoug