theAgingApprentice/HexaFloorRide #185: Add MQTT heakth message for high CPU usage

theAgingApprentice, 2023-03-12:


nerdoug, 2023-03-13:
Implemented this code in oneSec() at end of main():
```
      if( totCPU > 90)     // if the CPU is very busy, issue a red alert
      {
         TraceEf("CPU Utilization RED alert = usage% =",totCPU);
         CPULoad = CPURed ;   // note high CPU usage for load shedding purposes
         CPUOffload()         // try to reduce CPU usage somehow
      }
      else if (totCPU > 60)   // if the CPU is busy, issue a yellow alert
      {  
         TraceEf("CPU Utilization YELLOW alert = usage% =",totCPU);
         CPULoad = CPUYellow ; // note high CPU usage for load shedding purposes
         CPUOffload()         // try to reduce CPU usage somehow
      }
      else
      {
         CPULoad = CPUNormal  // we're comfortable - no action needed
      }
```
This issues the alert via an Error type trace, which may or may not go to MQTT, depending on trace configuration.

It may be desirable to issue alerts based on longer timeframes, such as "average CPU over 5 seconds exceeds 50%", although high average CPU over 1 second is severe enough to impact robot's ability to move.

nerdoug, 2023-03-13:
Information on CPU usage alerts has been added to the task-dispatch-monitor documentation file in section 2.8, Alerts Based on CPU Usage.

PR #192 (2023-03-13): Add MQTT health msgs / traces for high CPU usage
Fixes #185
commit 3f470464 (2023-03-13): Add MQTT health msgs / traces for high CPU usage
Fixes #185
closed 2023-03-13 by nerdoug