theAgingApprentice/HexaFloorRide #225: Build a remote control for the robot

theAgingApprentice, 2024-02-25:


nerdoug, 2024-03-10:
We decided to use a phone or tablet app rather than a new piece of hardware to control robot movement via generation of mqtt commands.

nerdoug, 2024-04-06:
current version of MQTT is 5.x . We use the older 3.1.1, which not all MQTT consoles support.

Browser based support:
MQTTX: web support is MQTT version 5.0 only?

Desktop:
MQTTX: 3.11. & 5.0, MacOS, Linux, Windows, open source
MQTT Explorer: 3.1.1 & 5.0, Windows, macOS, Linus

Mobile:
EasyMQTT: 3.1.1 & 5.0, iPhone, iPad, macOS
MyMQTT for Android: 3.1.1  & 5.0

I've downloaded this last app onto my phone. It doesn't support scripts, but does have a basic publish command.